uholeschak/ediabaslib #146: BMW ADS Inteface

dimiras, 2023-03-14:
Does this library make communication possible with BMW ADS modules without the need of an ADS interface?

uholeschak, 2023-03-14:
I think this will be only possible with a Deep OBD Bluetooth adapter.
Mybe very very old protocols are not implemented.

closed 2023-03-14 by uholeschak

dimiras, 2023-03-14:
Would it be possible to use an usb INPA cable?

uholeschak, 2023-03-14:
At least using an old ADS interface will create problems.
If your protocols don't require an ADS interface of course you could use an FTDI USB adapter.

dimiras, 2023-03-14:
I dont have an ADS Interface just a USB INPA cable like this one https://www.ebay.com/itm/185713725007
I want to use this on a BMW Z3 (E36/7) to program the OBC.
Do you think I can program it with this lib and cable?

uholeschak, 2023-03-14:
I think E36 is only using DS2, if you don't have a special ECU. Simply try.

dimiras, 2023-03-14:
Ok Thanks