uholeschak/ediabaslib #149: (Question / Feature Request) ELM327 Bluetooth-Security

micw, 2023-05-10:
Hello,

I have found this repo when I searched how I can secure my OBD2 adapter. From what I have read so far, most adapters are always pairable and many adapters use hardcoded bluetooth passwords which both are a security nightmare.
My adapter use case is to get battery data from my EV (nissan leaf), so I'd have the adapter always connected to the car. Is it correct that I can "simply" flash the alternative firmware from this repo to my adapter (or just by a pre-flashed from ebay) and set up a 16 digit pin to make the adapter secure? Does the firmware have all the functions the original firmware has?

Would you consider to add an option to your firmware to disable pairing after a certain amount of time (e.g. 10 mins after powering the adaper)? IMO together with a reasonable complex pin, this would improve security a lot.

Kind regards,
Michael Wyraz.

uholeschak, 2023-05-10:
If you use a Bluetooth chip that is programmable you could flash a corresponding firmware with PIC Kit to the PIC processor.
Perhaps the best solution would be for you, to connect the adapter power with the ignition, then the adapter is only reachable when driving.

closed 2023-05-10 by uholeschak

micw, 2023-05-10:
Unfortunately, I need the adapter when not driving (e.g. when it's in the garage and charging) because I need it to monitor the battery. I do it currently with the nissan API but it's very unreliable and only included for 3 years (which are over end of this year)

uholeschak, 2023-05-10:
Basically I sell adapters that have programmable Bluetooth chips. You could update the firmware to ELM327 V1.4 with the Deep OBD app. You have to change the PIN before updating the firmware!

micw, 2023-05-10:
Could you explain this a bit more? My understanding is that I get the ability to set a custom pin with your firmware - why/how should I set the pin before I flash the firmware?

uholeschak, 2023-05-10:
The ELM327 firmware is simply not able set the PIN, is has to be done with the custom firmware.

micw, 2023-05-10:
I see. And then I flash back the original? Is this required or does your firmware support the same features (primarily I need the feature to read the SoC state of the battery).
Will I be able to flash the custom firmware later after I have flashed the original on the device with Deep OBD?

uholeschak, 2023-05-10:
No my firmware is not ELM327 compatible only compatible with the Deep OBD app.
Yes you could flash back the custom firmware with the with Deep OBD app again.

micw, 2023-05-10:
> Yes you could flash back the custom firmware with the with Deep OBD app again

How does this work when I flash the original firmware? As far as I understand, the original FW does not have the bluetooth update option (otherwise, I would not need the initial PIC3 flash procedure), right?

uholeschak, 2023-05-10:
The EM327 has a patch, that allows to enable the bootstrap loader.

micw, 2023-05-10:
Alright, thank you very much. The device i have here has a ab22bp19227 chip (so no PIC). I'll send it back and order one of yours after I finished my tests.