uholeschak/ediabaslib #150: BK3211 compatibility

ChrisPHL, 2023-05-11:
Hi there,

I've bought an old ELM327 adapter years ago. The device looks very similar to the image posted here: https://github.com/uholeschak/ediabaslib/blob/master/docs/Replace_ELM327_HC04_Firmware.md
Looking into it reveals the Bluetooth chip is a BK32**1**1 (no typo, it's not BK3231).
Do you think it will be compatible, should I use the flash files within `/CanAdapterElm/bc04/`or better sue  `/CanAdapterElm/def*` from the Binaries-20230407.zip release archive?

kind regards
Chris

uholeschak, 2023-05-11:
Basically if you use the default firmware and the chip has the default baudrate it should work.

closed 2023-05-11 by uholeschak

ChrisPHL, 2023-05-11:
Thank you for your very fast (realtime! ;-)) support!