uholeschak/ediabaslib #152: Logging from multiple ECUs at once?

robot-army, 2023-06-21:
Hi, I'd like to log from the transmission, digital motor control, and battery management electronics on my F60 Hybrid. It seems to me that the app only logs the active 'view', as in, I can see log files for the digital motor control, but only when I have that tab active.

Is this expected behavior? Can someone let me know if I'm missing a setting or anything?

robot-army, 2023-06-21:
Interesting, OK, what's the cause of the performance hit from switching ECUs? Is it in the car, or in the software, or...both?

uholeschak, 2023-06-21:
The problem is, that a new ECU file has to be loaded every time when switching.
Then the complete initialization of the ECU file has to be executed.

robot-army, 2023-06-21:
Still not super clear to me, but I'm not at all familiar with the
architecture, either in the car or this software! (Thanks for this great
effort by the way, I'm amazed that it exists!)

If ediabaslib was somehow different, could ECU switching be streamlined?
Or, forgive my limited understanding, but does the ECU file get uploaded
from ediabaslib into the first device that the ENET cable sees, and that's
where the bottleneck is?

uholeschak, 2023-06-21:
There is only one communication channel and only one ECU file could be executed at the same time.
I would recommend to use an Ethernet connection to the vehicle for faster communication (ENET protocol).

robot-army, 2023-06-21:
Thanks, OK, yeah I'm using an enet cable to a small wifi access point,
which I then connect my phone to. I wish I could use a wired usb-c ethernet
connection to my phone, I feel like it'd be more reliable for sure.

uholeschak, 2023-06-21:
Perhaps if you use two smartphones and connect to different ECUs at the same time this could work, if the gateway could handle this in the vehicle (two TCP connections to the gateway).

closed 2023-06-21 by uholeschak

robot-army, 2023-06-21:
OK, an interesting idea, but the fact that you're proposing that says to me
that the performance bottleneck isn't inherent to the vehicle. What about
running two instances of Deep OBD? Or structuring Deep OBD so that each tab
in Deep OBD runs in a separate process, like web browsers?

Given that the datalog files are per-ecu, there wouldn't need to be any
competition for writing to the files. Architecturally, displaying gauges
from different ECUs on the same screen could be interesting, but each gauge
is only going to get data from one ECU, so...maybe not so bad?

uholeschak, 2023-06-21:
As you could see in the example you could write pages with multiple ecus.
The is the current solution.

robot-army, 2023-06-21:
Sure, thanks! Are there some instructions on how to get up and running with
android development for Deep OBD anywhere? I think I saw it for ediabaslib
but maybe not Deep OBD itself.

uholeschak, 2023-06-21:
For writing ecu pages you need only a text editor, then copy it to the device.

robot-army, 2023-06-21:
I was thinking of going deeper, and implementing a per-tab thread so it can
read from multiple ECUs without delay, but I won't be able to (it's a bit
of a pipe dream, as we say!)

uholeschak, 2023-06-21:
Install VS pro with Xamarin and start ...

robot-army, 2023-06-21:
OK! Will give it a shot! Thanks!

uholeschak, 2023-06-21:
If your are using the same app, you have to create multiple Ediabas instances at the same time.
But this is not not tested and will probably fail.
Perhaps multiple apps is a better solution ...