uholeschak/ediabaslib #155: Get rpm and set gpio output

gaucho1978, 2023-09-23:
Where should I start in order to use this firmware to periodically get motor speed in RPM and control a gpio pin in order to light a led above a threshold ? 
another basic question: can I use this firmware with my Car even if it Is not a BMW? I have a fiat alfa romeo Giulia

uholeschak, 2023-09-23:
May the test project is helpful:
EdiabasLib/EdiabasTest

closed 2023-09-23 by uholeschak

gaucho1978, 2023-09-23:
I think I was not clear. I would like to change the elm327 module firmware in order to work standalone by periodically checking motor speed and consequently power a local light. I don’t need to use external Microsoft windows tool like tue one you suggested.

uholeschak, 2023-09-23:
https://github.com/uholeschak/ediabaslib/blob/master/docs/Replacement_firmware_for_ELM327.md

gaucho1978, 2023-09-23:
Thank you mate, I’ve seen that firmware, and that Is the reason Why i posted here my question. I know Which AT command to send in order to get rpm speed. Do you know Which Is the firmware loop where i could insert the periodic request of rpm speed?

uholeschak, 2023-09-23:
Sorry this is not possible, every ECU required different requests.
I would recommend to use the app, read the data and log the communication.
The replay that.

gaucho1978, 2023-09-24:
Let me understand. From my computer i can send AT commands to the elm327. Correct?
i read that these are the commands:

AT D -> Set all to defaults
AT Z -> Reset Obd
AT E0 -> Echo off
AT L0 -> Line feed off
AT S0 -> Spaces off
AT H0 -> Headers off
AT SP 0 -> Set Protocol to 0 "Auto", search all protocols and connect it with proper protocol for that obd

010C -> Get motor speed in rpm

correct?

 Therefore the Bluetooth chip receives this message. Therefore I can inject my AT commands in the Bluetooth chip firmware. Correct? Does the Bluetooth chip modify the at command before to send it to the PIC18F25K80 chip? I think no. Therefore I can inject my AT commands directly inside PIC18F25K80 firmware. Correct?

uholeschak, 2023-09-24:
No this is not the way I am communicating with the the vehicle.
I think you should look for en ELM forum, you are wrong here.

gaucho1978, 2023-09-27:
I apologize. Your GitHub Is the only repository where i found FW for “Elm327”, this is Why i wrote here. Then if I Well understood your firmware Is inside folder /EdiabasLib/CanAdapterElm/CanAdapterElm.X, inside files “main” and “can” and it doesn’t implement AT commands. Is it correct?(question1)

 What Is instead contained inside folder /EdiabasLib/CanAdapterElm/CanAdapterElm.X/ELM327V15.X ? It seems the original firmware extracted From the chinese diagnostic tool, implementing AT commands too. Is it correct?(question2)

thank you mate

uholeschak, 2023-09-27:
Yes both are correct. Of course you are free to modify the ELM327V15.X source.