uholeschak/ediabaslib #162: How to use EdiabasTest.exe format

chaooe, 2023-11-15:
I am trying to use EdiabasTest.exe to run job, but i want to format the output of the job result.
I tried: 
`C:\EC-Apps\EdiabasTest\EdiabasTest.exe -f IGNITIONHISTORY=json -s PRG -j status_dlt_ip_configuration`
but the value of IGNITIONHISTORY is still the same as below
`IGNITIONCURRENT: I: -1 0xFFFF`

How to modify the format?

chaooe, 2023-11-15:
got the format from the code, using the regular express to filter them
[image]

closed 2023-11-15 by chaooe

uholeschak, 2023-11-15:
You could of course also modify EdiabasTest.exe if you want, adding new arguments ...

chaooe, 2023-11-15:
> You could of course also modify EdiabasTest.exe if you want, adding new arguments ...

Thanks, you are right. it would be prefect. i will have a try first.