uholeschak/ediabaslib #166: 3.3v available on FTDI chip of BMW DCAN cable but missing 100nF capacitor

Occupied, 2024-01-05:
Hi, maybe you've noticed this before but the design of the "famous" DCAN cable that uses the FTDI FT232RL chip has a design mistake, quoting from the datasheet here below.
Before adding the recommended capacitor to the 3V3OUT pin there is ca 400mV of ripple on it, after adding the cap there is no ripple.
I'm posting this here as I thought maybe you would find it useful for your documentation.
Adding the capacitor should make the USB transfer more reliable.
The additional positive is that it could be used to directly power a 3.3v bluetooth regulator if it draws a maximum of 50mA.

> Name: 3V3OUT Type: Output
> +3.3V output from integrated LDO regulator. This pin should be
> decoupled to ground using a 100nF capacitor. The main use of this pin is
> to provide the internal +3.3V supply to the USB transceiver cell and the
> internal 1.5kΩ pull up resistor on USBDP. Up to 50mA can be drawn from
> this pin to power external logic if required. This pin can also be used to
> supply the VCCIO pin.
[image]

uholeschak, 2024-01-05:
Thanks for the info.
Currently the state of the cable is "depreciated" and I don't make any more changes to the software.
Probably this depend form where you buy the cable. Do you have any source specification? How to identify this cable type?

uholeschak, 2024-01-05:
I have now updated the D-CAN docs.

closed 2024-01-05 by uholeschak

Occupied, 2024-01-05:
Thank you!
From what I could find every one of the BMW K+DCAN cables have the same design mistake.
(It is easy to count the SMD caps on the top and bottom, on mine they all supply +5v and all I could find pictures of online look very similar).
There is another mistake with the selection of R5 & R6 that is the voltage divider for the ignition detection that go to Atmega pin PB4.
OBD pin 1 go to R6 1K and then the Atmega PB4, in paralell with this is R5 47k to ground.
This creates a voltage divider that reduces the 12v to 11.75v, the absolute maximum rating is 5.5VDC for the Atmega.
I guess it only works because there is some protection in the Atmega, I measured over 5.8v on the pin.
On mine I moved the 47k resistor to R6 and then replaced R5 with a 18k resistor.

uholeschak, 2024-01-05:
Thanks for the info. I expect the Atmega has some diodes for over voltage protection.

Occupied, 2024-01-05:
Yes they do, I found a paper from Atmel (doc2508.pdf) that recommends a max of 1mA of current, with the setup above (1k resistor to +12v) ohms law tells us that is 12mA, I guess there is some headroom because mine still lives ;)
Anyway, I just wanted to add that you should probably put the name of the chip (FTDI FT232RL) in the updated documentation so it is clear what is referenced.

uholeschak, 2024-01-05:
Ok thanks I have updated the docs.
If the motor is running the OBD voltage is much higher, near 15V. Don't forget that when calculating the resistor.

Occupied, 2024-01-06:
Yes I know, the 47k/18k resistors I choose work from around 11v (Atmega sees 3v) up to about 18v (Atmega sees 5v).
Only BMW's with a K-Line interface use the ignition input (OBD wire 1 is missing in cars that switched from K-Line to Canbus).