uholeschak/ediabaslib #169: Connect to ESP8266 Deeb OBD ELM327 through local network

shar34, 2024-02-28:
I have Android multimedia with USB WiFi modem connected and connected Deep OBD ESP8266 adapter to modem network. I can ping adapter through local network, but in application button "Connect" not available until I connect to a WiFi network or turn on the access point, after this (enable access point for example), button get available and I can connect to adapter with IP address in usb0 local network (no matter if there any clients connected to access point or not):

```
console:/ # ifconfig usb0
usb0      Link encap:Ethernet  HWaddr 6e:29:65:dd:24:bb  Driver cdc_ether
          inet addr:192.168.1.200  Bcast:192.168.1.255  Mask:255.255.255.0 
          inet6 addr: fe80::6c29:65ff:fedd:24bb/64 Scope: Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1494 errors:0 dropped:0 overruns:0 frame:0 
          TX packets:1458 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:135804 TX bytes:139198 

console:/ # ping 192.168.1.34 # <- Deep OBD address in usb0 local network
PING 192.168.1.34 (192.168.1.34) 56(84) bytes of data.
64 bytes from 192.168.1.34: icmp_seq=1 ttl=128 time=227 ms
64 bytes from 192.168.1.34: icmp_seq=2 ttl=128 time=45.3 ms
64 bytes from 192.168.1.34: icmp_seq=3 ttl=128 time=67.6 ms
64 bytes from 192.168.1.34: icmp_seq=4 ttl=128 time=88.9 ms
64 bytes from 192.168.1.34: icmp_seq=5 ttl=128 time=2.66 ms
64 bytes from 192.168.1.34: icmp_seq=6 ttl=128 time=2.74 ms
```

Is possible to make "Connect" button available with no WiFi or AP on (maybe add some "force" or "local network" options in settings)? thanks.

uholeschak, 2024-02-28:
You could enter a manual IP address in the ENET IP dialog field. This is only possible if the adapter is an ENET adapter.

closed 2024-02-28 by uholeschak

shar34, 2024-02-28:
With no WiFi or AP enable this settings not available (I have Deep OBD ELM327 adapter with ESP8266 for E-series).

uholeschak, 2024-02-28:
If you select the Deep OBD WIFI adapter type you could also enter an adapter ip manually.

uholeschak, 2024-02-28:
LAN connection has many problems and should not be used.

shar34, 2024-02-28:
[image]
[image]
[image]

But adapter connected at this moment.

shar34, 2024-02-28:
[image]
Just enable access point and I can connect to adapter in LAN network.

uholeschak, 2024-02-28:
So it is solved? Are us using hotspot mode now?

uholeschak, 2024-02-28:
What is the problem? Simply enable WiFi. Only one click.
This is a singular problem only required for one person.

shar34, 2024-02-28:
Problem with android auto, when connect phone, it automatically enable\disable WiFi and hotstop, in this case, adapter get disconnected, and even enable WiFi and not connect to WiFi network, button still disabled. I love this app and I've been wanting to send a donation for a long time, but unfortunately I can't do it through PayPal.

uholeschak, 2024-02-28:
The question is, if connected to android auto is the routing is still working? The problem is, that the data is bound to an interface connection type so it may fail anyway.

shar34, 2024-02-28:
Yes, check now, I can connect to adapter while android auto connection working.

uholeschak, 2024-02-28:
Ok so a global option to disable the "WiFi is enabled" check would be sufficient?

shar34, 2024-02-28:
Yes, thanks very much. I'll try to found someone who has euro card to send donation through PayPal.

shar34, 2024-02-28:
It's seems like donation send successful (around 20-21 euro). Thanks for the quick replies and help.

uholeschak, 2024-02-28:
Thanks, maybe I could also detect if the specified IP matches a valid Ethernet connection. Similar code must exist for the ENET interface.

uholeschak, 2024-02-29:
I have now uploaded a new beta active with:
https://play.google.com/apps/testing/de.holeschak.bmw_deep_obd
In this version the local Ethernet interface is detected in DeepOBDWifi mode. If you set an adapter IP address and it **matches the subnet of the Ethernet interface**, then the button is enabled. The check for a valid adapter is also disabled in this mode.

shar34, 2024-03-01:
In last beta version buttons still disabled ("Connect" button too), but adapter connected to local 'usb0' network.
[image]

uholeschak, 2024-03-01:
As described, the important point is that the network adapter is configured correctly with a subnet and the selected ip address is inside the range of the subnet. In the edit dialog for the ip address the adapter and the subnet is displayed, if WiFi is off.

shar34, 2024-03-01:
```
127|console:/ # ifconfig usb0 
usb0      Link encap:Ethernet  HWaddr 36:20:56:28:26:3c  Driver cdc_ether
          inet addr:192.168.1.200  Bcast:192.168.1.255  Mask:255.255.255.0 
          inet6 addr: fe80::3420:56ff:fe28:263c/64 Scope: Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2570 errors:0 dropped:0 overruns:0 frame:0 
          TX packets:2782 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:250410 TX bytes:263554 

console:/ # ip route
192.168.1.0/24 dev usb0 proto kernel scope link src 192.168.1.200 
console:/ # ping 192.168.1.34 
PING 192.168.1.34 (192.168.1.34) 56(84) bytes of data.
64 bytes from 192.168.1.34: icmp_seq=1 ttl=128 time=2.85 ms
64 bytes from 192.168.1.34: icmp_seq=2 ttl=128 time=2.59 ms
^C
--- 192.168.1.34 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 2.592/2.722/2.853/0.140 ms
console:/ #
```

On Android headunit I set 'usb0' adapter to static 192.168.1.200 address (192.168.1.1, 255.255.255.0).
Adapter also set to static address - 192.168.1.34. And I can get adapter settings from browser.
Maybe DeepOBD application not see this 'usb0' local interface?

uholeschak, 2024-03-01:
The adapter must official visible in android. It has to be a valid official ethernet adapter. There must a Ethernet configuration page in the android system settings.
What is displayed in the edit dialog for the ip address? Is there adapter ip and subnet displayed (screenshot)?

shar34, 2024-03-01:
This network interface is missing in the Android system settings. No not any notifications while plug, I do manually bring up and set address from terminal and can access devices in this local network.
[image]

uholeschak, 2024-03-01:
So this means in the ip edit dialog of the app no adapter information is displayed?

uholeschak, 2024-03-01:
Could you make a screenshot of the ip edit dialog in the app (WiFi off)?

shar34, 2024-03-01:
Connected standard USB Ethernet adapter and connect to some router, 'eth0' configured automatically (I don't execute any commands in terminal), new icon appear in status bar, but still no any ethernet adapters in android system settings and edit IP in DeepOBD still disabled. 
[image]

uholeschak, 2024-03-01:
If you have custom rom they may have removed Ethernet support in Android because it's not required ...
If you use a smartphone and attach an Ethernet adapter, an icons in the toolbar appears to configure the adapter.
If you click this item you get into the android system settings for the adapter and could specify the ip address.

shar34, 2024-03-01:
Any solutions (the easiest to implement in code) to make "Connect" button enabled without WiFi/Hotspot/Ethernet will works.

uholeschak, 2024-03-01:
And the check if a valid adapter is connected must be disabled ...

uholeschak, 2024-03-01:
I have published a new beta, it requires some time to update.
In the global settings to go **export configuration**. The file location is displayed. In this file change the XML node `DisableNetworkCheck` at the end of the file to `true`. Then use **import configuration** in the global settings. Afterwarts the button and the menu entry should be always enabled.