uholeschak/ediabaslib #173: deepObd on laptop

windev92, 2024-05-13:
Hello Ulrich,

I would like to run the app on desktop (no android).
It is written in C# so probably possible to build it with VS without too much of work I guess.
Do you have it running on pc too?
Also do you have a test app that sits on top of ediabaslib that runs on PC?

Thanks

uholeschak, 2024-05-13:
Hello,
sorry this is not easy to implement. The app is only for implemented for Android.
If you want to do it, no problem it's open source.
Only the Ediabas core is portable.
The solution is, to use it with ISTA-D.

closed 2024-05-13 by uholeschak

windev92, 2024-05-13:
But ista-d is not open source. So I cannot understand things easily. Do you test the lib on android or do you have something running on pc?
Thanks

uholeschak, 2024-05-13:
You could try it with the test tools:
https://github.com/uholeschak/ediabaslib/tree/master/EdiabasLib/Test/EdiabasCall
https://github.com/uholeschak/ediabaslib/tree/master/EdiabasLib/Test/EdiabasLibCall