uholeschak/ediabaslib #174: Readme not up to date?

windev92, 2024-05-13:
Hello Ulrich,

The readme here https://github.com/uholeschak/ediabaslib/tree/master/BmwDeepObd

is telling to rename things in the Config directory, but there is no Config directory
Bests

uholeschak, 2024-05-13:
Thanks for the info, the file is old and garbage. Deleted now.

closed 2024-05-13 by uholeschak

windev92, 2024-05-13:
So it means we don't need to configure anything regarless of the car?
Cause the app works a treat! I left it opened at home, then put an otg cable and plugged it to the car. The app detected in the background that it is usb and needs to be using ftdi with a notif!
Then it worked the first shot and got the VIN!

uholeschak, 2024-05-13:
The file was a hint that the sample files are for one specific vehicle only. The correct way is to use the configuration generator.