uholeschak/ediabaslib #175: Bestdis input

windev92, 2024-05-15:
Hello,

I would like to run BESTDIS tool executable to see how things work in real.
It requires as input a bestobjfile and the code searches for other files in the same directory.
Can you please add in BESTDIS direcrory a kinda test directory with the minium files needed so that we can run it and see the output files that are generated. I also see that  .grp and not .grp files have different outputs.
Thanks

uholeschak, 2024-05-15:
You could use the .prg and .grp files from INPA or ISTA.
The files are copyright by BMW and could not be added.

closed 2024-05-15 by uholeschak

windev92, 2024-05-15:
Ah I understand.
it needs a bestobjfile as input. So I need to pass in a prg or grp as argv(0) ?

uholeschak, 2024-05-15:
Yes

windev92, 2024-05-15:
It lools like there is a problem here:

If you pass in : a/b/f.prg, the code below will search for 
"f.prg" in "a/b" so the loop will be useless as it will always be 1 element, i.e argv(0)

string[] files = Directory.GetFiles(dirName, Path.GetFileName(args[0]));
            foreach (string file in files)

uholeschak, 2024-05-15:
Thank I have adapted the hint test. Must be a directory.

windev92, 2024-05-15:
Any idea where I can get more information about what is the difference between grp and prg file, and if there is a tool that shows how they are generated from the source code(it seems to be a compiler)
thanks

uholeschak, 2024-05-15:
Basically here is a wrapper for the BEST Compiler, but it will only work with EDIABAS 7.6.0
https://github.com/uholeschak/ediabaslib/tree/master/Tools/BestNet
Otherwise use Best1.exe or Best2.exe that comes with EDIABAS. Finding the documentation is up to you.