uholeschak/ediabaslib #178: Connect EdiabasLib:6801

aosbmw, 2024-06-13:
How can i do it? 
Start Ista do diag + open nsc expert for coding (like interface=remote, port = 6801 in ediabas.ini)

uholeschak, 2024-06-13:
Sorry I don't understand what is your problem. The configuration is possible in the EdiabasLib.config file:
https://github.com/uholeschak/ediabaslib/blob/master/docs/EdiabasLib.config_file.md
If you are using a Deep OBD adapter you should use the EdiabsLibConfig tool. This will configure the file correctly.

closed 2024-06-13 by uholeschak

aosbmw, 2024-06-13:
I want use NETMAN server ifhsrv32.exe with EdiabasLib.  But if i replace orig dll api32.dll
ifhsrv32.exe not start.

uholeschak, 2024-06-13:
Use an https://github.com/uholeschak/ediabaslib/blob/master/docs/ENET_WiFi_Adapter.md#alternative-wifi-adapter and the Config tool https://github.com/uholeschak/ediabaslib/releases