uholeschak/ediabaslib #179: Instructions for flashing with a different programmers

TzOk83, 2024-06-14:
Original description states only that you need to read the original hex first, but what is the reason for it, what you need to do with the original hex file? I don't have PICkit, but I have T48, which supports PIC18F25K80. I guess it is about saving User ID or some configuration bits, but which of them should I keep? What if I already have the empty chip?

uholeschak, 2024-06-14:
The unified.hex file is for first programming only and contains the full image including the boot loader.
The other file is for firmware update only by the Deep OBD app. The boot loader is missing in case of course.

closed 2024-06-14 by uholeschak

TzOk83, 2024-06-14:
If the unified.hex is a full image, then what for do I need to read the original hex from the device?

uholeschak, 2024-06-14:
Where did you find that info? Perhaps I should simply remove this text to avoid questions ...

TzOk83, 2024-06-14:
It is in "Replace_ELM327_HC04_Firmware.md" in Step 2, bullet 3. You won't get a proper HEX file, because PIC is read-protected, so it reads as FFs.

Either way I can't make it work on my old transparent-blue cased ELM327, based on PIC18F25K80 and BK3231.

uholeschak, 2024-06-14:
Thanks, I have removed the sentence. It makes no sense. This chapter has been written by somebody else.

TzOk83, 2024-06-14:
Ok, so I have flashed the default variant and it started to work, thanks.