uholeschak/ediabaslib #182: FTDI adapter stopped working after recent update - not sure where to look.

martinroger, 2024-08-11:
Hi, I need a bit of help to understand why DeepOBD doesn't seem to detect anymore my D/K-can "generic" INPA USB cable. Until recently, it was working fine with DeepOBD on my phone (Pixel 5, latest available Android) but today it just didn't seem to detect the adapter.

The adapter appears in the USB management notifications on the phone, and for what it's worth is recognized by Protools.

I tried deleting the cache and storage of the app and restarting fresh, alas it doesn't seem to help. I also see the custom BT adapter isn't available for purchase lately. Any clue where I could be looking for an obvious issue ?

uholeschak, 2024-08-11:
Maybe you have assigned the adapter permanent to another application, in this case you have to release if from the other application first.
You could buy the BT adapter here: https://www.ebay.de/itm/256599171440

closed 2024-08-11 by uholeschak

martinroger, 2024-08-11:
I think I found how to. I'll try unassigning it from the other app (likely Protool) and see how that goes. Thanks !

S0PEX, 2024-08-22:
@uholeschak Somewhat off-topic, but would the Wi-Fi based Adapter be better in terms of stability and speed?

uholeschak, 2024-08-22:
That depends of course of the WiFi stability ...
Basically USB has problems with the ground noise from ignition in the vehicle.

S0PEX, 2024-08-22:
If you had to choose an adapter, would you go with BT or Wi-Fi? From what I gather, BT is the recommended option?

uholeschak, 2024-08-22:
There is no general answer.
For ENET based vehicles WiFi makes more sense, because it is faster.
Bluetooth is preferred für CAN.

S0PEX, 2024-08-22:
Is there a reason to use WiFi for an ENET vehicle? I have an E87, so it uses CAN, but I would think that the higher bandwidth for a CAN vehicle wouldn’t be a disadvantage.

uholeschak, 2024-08-22:
CAN ist not so really fast. In this case Bluetooth is sufficient ...