uholeschak/ediabaslib #183: MHD/Thor WiFi with API32.dll

Vernerio, 2024-08-12:
Hi, API32.dll Releases after 20230108 not connect with MHD.
This works with older DLL's.

"
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
    <appSettings>
        <add key="Interface" value="STD:OBD"/>
	<add key="ObdComPort" value="DEEPOBDWIFI"/>
	<add key="ObdKeepConnectionOpen" value="0"/>
    </appSettings>
</configuration>
"

uholeschak, 2024-08-12:
What is an MHD? Ist this an ELM adapter? This adapter working with the App?
If this is a fake ELM use an ELM with original firmware.

Vernerio, 2024-08-12:
https://mhdtuning.com/pages/mhd-wireless-adapter
https://bimmer-connect.com/products/bg-thrwi

MHD & Thor adapter IP is 192.168.4.1
Here is some discussion about MHD or Thor with EdiabasLib. 
https://www.e90post.com/forums/showthread.php?t=1745311

uholeschak, 2024-08-12:
At least this adapter has not the replacement firmware. So `DEEPOBDWIFI  `is incorrect. Is this is an ENET adapter use the `ENET` setting.

closed 2024-08-12 by uholeschak

gardar, 2024-08-21:
It's a enet adapter and it works great with deep obd I just had to choose `ENET / Deep OBD WIFI` and the gateway IP `192.168.4.1`. Also using it with ISTA / E-Sys.

[image]

uholeschak, 2024-08-21:
Thanks for the info. `ENET` seems logical for this adapter type.

alex77272, 2024-11-18:
I also have problems getting the THOR flasher to work with ISTA 

https://de.bimmer-connect.com/products/bg-thrwi

They also provide an api32.dll and an ediabaslib.config file. When I run an identification test on ISTA, it does not fail immediately, but tries for 30 seconds and then fails. 

What I am also wondering about in the ediabaslib.config:

<add key="ObdComPort" value="DEEPOBDWIFI"/>.

Shouldn't the value be THOR_Wifi? The name of my wifi? If I do that, the test fails immediately. 

INPA works.

uholeschak, 2024-11-18:
You have not attached the config file, but you don't have to specify the WiFi network.
You simply must be connected with the network.

alex77272, 2024-11-18:
OK, I am connected to the WiFi network. I also don't understand why INPA works but ISTA doesn't. I have replaced the api32.dll in the rheingold/ediabas/bin folder and added the EdiabasLib.config. 

[config.zip](https://github.com/user-attachments/files/17805824/config.zip)

uholeschak, 2024-11-18:
I don't know this product, so the correct solution is to contact the THOR developer.