uholeschak/ediabaslib #184: Gauge.S support for ediabaslib

handmade0octopus, 2024-08-14:
Hello!

I am creator of device that could be found there: http://gauge.sorek.uk/

I wish to support DeepOBD through WiFi/BT (prefer the first one). If you could let me know where I could find some info on implementation on my side it would be super cool. Device has k-line and CAN bus and it's based on ESP32 so it should be doable to do super speeds with proper function. MHD/xHP flashing could also be doable with this approach.

If you have any pointers please let me know.

Kind regards,
sorek
contact@sorek.uk

uholeschak, 2024-08-14:
Basically Deep OBD has a broadcast interface:
https://github.com/uholeschak/ediabaslib/blob/master/docs/Page_specification.md#broadcasts
You could create an app that uses that interface and is gateway to your device.

handmade0octopus, 2024-08-14:
But that wouldn't work with api32.dll correct?

I want to emulate drop in solution for all existing ones. I know AT type commands from ELM327 are common but is there a better protocol I am not aware of?

handmade0octopus, 2024-08-14:
I think you missed my point.

I want my device to be accepted by api32.dll and DeepOBD - so I need to implement protocol inside the device so it communicates with it well.

I could use AT commands and mimic behaviour of ELM327 but I wonder if there is better way than that. I need to implement it by hand so I am not bound by having to stick to AT type commands - I just need to know which one is most common and woks best.

uholeschak, 2024-08-14:
Maybe you should create a block diagram of your intended design first. If you want to use ediabas lib you need of course a complete ediabas installation.

handmade0octopus, 2024-08-14:
????

I just want to create Firmware for my device that will mimick ELM327 or other adapter with it's protocol.

I am looking for pointers if there are any code sources you can recommend so I can add support to my device so it works like just another supported adapter. I want to choose the fastest and most popular one.

uholeschak, 2024-08-14:
Maybe you could check this file:
https://github.com/uholeschak/ediabaslib/blob/master/EdiabasLib/EdiabasLib/EdElmInterface.cs
It implements the BMW-FAST protocol over ELM327.

closed 2024-08-14 by uholeschak

handmade0octopus, 2024-08-24:
> Maybe you could check this file: https://github.com/uholeschak/ediabaslib/blob/master/EdiabasLib/EdiabasLib/EdElmInterface.cs It implements the BMW-FAST protocol over ELM327.

Dear @uholeschak 
If I wanted to implement custom adapter protocol into my adapter - how would I approach that? What are `82 F1 F1 FE FE 60` bytes mean (I know last one is checksum) and how should I approach each one?

Is there a list or document that will help me write custom firmware for my adapter? I would rather not use AT ELM327 style but go for ENET adapter with proper custom FW.

Thank you!

uholeschak, 2024-08-24:
The adapter source cold be found here:
https://github.com/uholeschak/ediabaslib/blob/master/EdiabasLib/CanAdapterElm/CanAdapterElm.X/main.c
There is no explicit documentation. Read the source ...