uholeschak/ediabaslib #185: VAG, ECU found, ecu connected, configured for engine, configured first 4 bits of data on channel 1 but no data returned when connected

chapelhillmccl, 2024-08-18:
HI.
Apologies if this is a user error, I have tried numerous times and re-read most of the documentation but seem to be stuck.
Using dongles based on "Replacement firmware for ELM327", tested working with BMW E60 and I3.
Trying to connect to Skoda Fabia 1.9 diesel 2007 PD engine.
From Deep OBD, Selected Manufacturer Skoda
Ecu file downloaded and unpacked via Deep OBD android app
Using recent version Deep OBD 1.3.229
Detect ecu completes and finds at least 1 ecu including Engine Electronics,
Generate configuration, select Engine electronics, add first 4 data lines, all on channel 001.
(Ecu header info can be seen as would also be seen using Rosstech VCDS)
Save and use as current configuration.
Connect to adaptor
The 4 lines show, but with no data returned, no numbers nothing.

Not sure if there is something else VAG specific that I am missing.
Tried two different dongles.
Tried Scanning as VW, found only Engine electronics, same results in configuring the 4 channels.

Error page seems to scan the engine module and gets to 100% reports no errors.
Updated dongle to ELM firmware and then back to Deep OBD firmware via Deep OBD app.

Any assistance from anyone would be most welcome.
Thanks

uholeschak, 2024-08-18:
Basically it's possible, that the returned ecu data is simply invalid or the ecu is not reporting the information.
You could also check this by using the read button on the ecu page.

If I want to check this I need foll set of trace files:
- Enable collecting debug data in the global settings
- Execute the complete configuration generator process (Delete all old configurations before!) from the beginning and also open the ecu pages.
- Send the trace file from the configuration generator menu. Specify your mail address or similar during sending and report it here
- Start communication on the main page and open all important pages.
- Stop the communication on the page and send the trace file from the main menu.

uholeschak, 2024-08-18:
No results for the read button is at least consistent. So no result on the main page is at least correct.
If you you select other results, do you get valid values in this case?

chapelhillmccl, 2024-08-18:
I will double check when the car comes back.
 Thanks

chapelhillmccl, 2024-08-18:
I have sent two traces alastair.mcclureATgmail.com
I will not get access to the car until next Saturday.
No problem if you don't have enough time.

Deleted everything, uninstalled and re-installed etc..

I did find a setting in Global preferences to use old VAG mode, but that does not seem to work with the current VAG files, but did not have enough time to allow full scan, juts enough to find engine electronics.
Thanks
Regards

chapelhillmccl, 2024-08-18:
[image]

uholeschak, 2024-08-18:
Thanks I found the log files, but the data is not enough to regenerate the configuration without a full scan.
Maybe you could zip the complete configuration for the smartphone and attach it.
The old VAG ist not useful any more. Do not activate it. Maybe I should remove it.

uholeschak, 2024-09-06:
Response timed out.

closed 2024-09-06 by uholeschak