uholeschak/ediabaslib #186: Coding of ADS ECUs (ASCMK4) failed

hell046, 2024-09-16:
I was happy to see that with an deepOBD Bluetooth OBD adapter I was able to read older ECU's of a e36 that require ADS protocol. Reading faults and everything else with INPA worked perfectly for all ECU I tried. 

My goal was to do coding on these ECUs as my ECU was broken and I have to adapt a new ECU to my car. Most ECUs worked, but unfortunately for the ABSMK4G1 I was  not able to code it successfully. 

If I try to read the coding of the ECU I get the message you can see in the screenshot. 

[image]

Do you have an idea what the problem is? All other ECU worked perfectly!

uholeschak, 2024-09-16:
Sorry I have never used NCS with EdiabasLib, so I have zero information.
Maybe could generate a ifh.trc api.trc file and check what exactly is the reason for the problem.

closed 2024-09-16 by uholeschak

hell046, 2024-09-16:
I cant imagine it is not responing. When coding sometimes NCS then reads the coding of all ECU and I saw the values of the bespoked ASC ECU. Therefore it seems to be able to read the coding. 

How can I do ifh.trc and api.trc?

uholeschak, 2024-09-16:
You could try with in EDIABAS\Bin\EdiabasLib.config
```
    <add key="LockTrace" value="1" />
    <add key="ApiTrace" value="1" />
    <add key="IfhTrace" value="2" />
```
The traces file are in EDIABAS\Bin\Trace
Delete old trace before the test.

hell046, 2024-09-16:
Thanks for the information, I will try it when Im back at the car (it can take a while as the car is elsewhere right now). 

I will keep you updated!

uholeschak, 2024-09-17:
Thanks for the info. I could not test it because I don't have so old vehicles. An simulating is difficult.

hell046, 2024-11-06:
Hi, I found some time to do some Traces. I did one with INPA reading out the ASC which works perfectly. Then I did one with NCS trying to code the ASC which did not work as I already said. Then I tried to read the coding of the instrument cluster which also does require ADS diagnostics and this worked fine without a problem.

Maybe you can see whats the problem with the Traces. 
[Trace Files E36.zip](https://github.com/user-attachments/files/17652698/Trace.Files.E36.zip)

uholeschak, 2024-11-06:
Is the problem in ASC NCS Expert Codierung lesen?
At the end there is an error:
```
apiJob('C_ASCT', 'IDENT', ''=[], '')
apiResultText('JOB_STATUS', 1, '')
=True ('OKAY')
=([4F 4B 41 59 ])
apiResultText('ID_SW_NR', 1, '')
=True ('B4')
=([42 34 ])
apiJobData('C_ASCT', 'C_C_LESEN', [01 01 00 00 00 00 00 00 00 FF 00 00 00 06 00 06 00 0F 00 00 00 FF FF FF 00 FF FF 03 ], 28, '')
apiResultText('JOB_STATUS', 1, '')
=True ('ERROR_ECU_NACK')
=([45 52 52 4F 52 5F 45 43 55 5F 4E 41 43 4B ])
apiState()=1 busy=0
apiEnd()
```
Basically the communication is ok:
```
 (Send): 36 00 05 00 33 
UpdateAdapterInfo: ForceUpdate=False, VoltageUpdate=False
 (AdSend): 82 F1 F1 FE FE 60 
 (AdResp): 82 F1 F1 FE FE 60 82 F1 F1 FE 81 E3 
 (AdSend): 84 F1 F1 06 55 AA D5 40 
 (AdResp): 84 F1 F1 06 55 AA D5 40 84 F1 F1 06 55 AA D5 40 
 (AdSend): 82 F1 F1 FD FD 5E 
 (AdResp): 82 F1 F1 FD FD 5E 85 F1 F1 FD 00 03 00 0F 76 
 (AdSend): 82 F1 F1 FB FB 5A 
 (AdResp): 82 F1 F1 FB FB 5A 89 F1 F1 FB 98 E7 76 84 67 A2 D2 85 3F 
 (AdSend): 82 F1 F1 FC FC 5C 
 (AdResp): 82 F1 F1 FC FC 5C 82 F1 F1 FC 7B DB 
IgnitionStatus: 81
EscapeModeRead: False
EscapeModeWrite: False
AdapterType: 3
AdapterVersion: 0.15
AdapterSerial: '98E7768467A2D285'
AdapterVoltage: 12.3
 (Resp): 36 00 0B 00 00 00 B4 00 00 00 89 
```
Is this the location with the error?

hell046, 2024-11-06:
Thats correct, you found the correct location.

The Problem is I dont have an native ADS Interface which is needed to communicate with the ASC control module. Therefore I cant compare with working files. 

In my opinion it should be the same as the communication with the KMB in the other Folder. Same thing there, you need an ADS Interface and there your library works. I just dont get why it refuses to code ASC.

uholeschak, 2024-11-06:
There are two possibilities:
* The request telegrams are incorrect
* The response telegrams are correct and misinterpreted.
I expect the last. Communications seems to be ok, but the result is ERROR_ECU_NACK.