uholeschak/ediabaslib #189: Any plan to support the USB ELM327?

Conti93, 2024-11-21:
As per title, I have an ancient wired ELM327 adapter, which to be fair I could just yank the MAX232 out of it and put an HC05 module in but... I don't really like bluetooth on PC.

If that's out of the question I'll search for a bluetooth module/ELM.
Thanks.

closed 2024-11-21 by uholeschak

Conti93, 2024-11-21:
> No this is not planned. USB is unstable in the vehicle, because of mass loops.

I was writing an EDIT, but you were faster lol. Btw: I'm a fucking idiot. I was thinking about the HC05, once connected it acts fully transparent so it should already work as an ISTA USB interface, given that I use a laptop disconnected from ground, right?

uholeschak, 2024-11-21:
There is a COM port support in EdiabasLib. But only for standard USB Adapters mit Atmel processor.

Conti93, 2024-11-21:
Isn't the HC05 used by the Deep OBD adapter also recognized as a COM interface?
[image]