uholeschak/ediabaslib #191: VW Passat B5 1.8 125hp 1997

VasilKotsov, 2024-12-04:
Hello I have a plans to buy ELM327 and I don't know if Deep OBD will work with my car. I think my car support  KW 1281 protocol.

uholeschak, 2024-12-04:
VAG ist not supported for ELM. You have to use the replacement firmware.
https://github.com/uholeschak/ediabaslib/blob/master/docs/Replacement_firmware_for_ELM327.md

closed 2024-12-04 by uholeschak

VasilKotsov, 2024-12-04:
So if I buy this adapter https://www.aliexpress.com/item/33059701156.html?spm=a2g0n.productlist.0.0.22873e4cqhUuqs&browser_id=885d34242b3f47d28785b024300d7fe2&aff_trace_key=ef1b60afa041411890d8f1d7835bd1d3-1733046924818-09884&aff_platform=msite&m_page_id=mfphvvzlscawrusb19391ca79f31e23da26f1c94b3&pdp_npi=4%40dis%21BGN%215.92%215.92%21%21%213.12%213.12%21%402103854617333177860798050e2213%2112000027641196444%21sea%21BG%212726824987%21X&algo_pvid=6dfd3280-eb69-47af-a44e-4d0c1000244c
And flash replacement firmware it will work.

uholeschak, 2024-12-04:
You have to make sure this is a real PIC processor no fake. The price is to low for this.
You could also buy a preprogrammed adapter: https://www.ebay.de/itm/256599171440

VasilKotsov, 2024-12-04:
Okey,. Thank you. I will buyin one.

VasilKotsov, 2024-12-21:
Hello again. I've I bought your preprogrammed adapter from EBay. But I also have this from AliExpress and I want to reprogramme it. 
What is the easiest way to reprogramme it? 
[image]

VasilKotsov, 2024-12-21:
[image]

VasilKotsov, 2024-12-21:
![Uploading IMG_20241221_165028021_HDR.jpg…]

VasilKotsov, 2024-12-21:
[image]

uholeschak, 2024-12-21:
You have to use the PicKit programmer and connect it to the correct programming pins.
You have to find you the correct wiring yourself.  Every PCB is different. Read the data sheet of the processor for this.