uholeschak/ediabaslib #196: Deep OBD with New Beatle and values

Weljemi, 2025-02-11:
Deep OBD with New Beatle 2007, SIMOS33 1.6l ECU. I have selected Engine electronics and two values to be displayed. The values show up as:

001/2 Coolant       3.000000E+000
004/2 Basic Functions: Voltage Supply  1.262500E+001

The values are correct, 3 celsius and 12,62 volts. Could this be fixed and made human readable?

uholeschak, 2025-02-11:
Basically I have to simulate the car communication for this.
The best way to this, is to completely scan and configure the car from the beginning and send the resulting traces files manually from the configuration generator menu.
Additionally you have to attach a video or screenshots, which items you have selected.

Another option would be to if you zip and attach the complete generated configuration.
Then start communication on the main page and send the traces manually after stopping the communication.

Weljemi, 2025-02-11:
How do i zip and attach config?

uholeschak, 2025-02-11:
Go to app directory ...\de.holeschak.bmw_deep_obd\files\Configuration and locate the correct configuration directory.
Then copy the directory to a PC and zip it manually.

Weljemi, 2025-02-11:
I sent the trace via the app. Here is the config.
[Manual1.zip](https://github.com/user-attachments/files/18759410/Manual1.zip)

uholeschak, 2025-02-11:
If I see it correctly you have a selected gauge display with mode text?
I think this makes no sense.
Why not simply disable gauge display and use it as list?
In `format=""` tag nothing is specified. This will of course generate default output.
Maybe something like `format="6.2R"` ?
https://github.com/uholeschak/ediabaslib/blob/master/docs/EDIABAS_result_types_and_formats.md

Weljemi, 2025-02-11:
If Gauge means graphical the checkbox is not checked. Maybe I should clear the config and create new, I have had the app for a while.

If I select each of the values and "Output format", User, 6.2R it works. I get 1.50 and 12.63.

uholeschak, 2025-02-11:
Sorry my mistake, `display-mode="list"` is set. So this is ok.

closed 2025-02-11 by uholeschak