uholeschak/ediabaslib #198: OBDLink MX+

einfacharthur, 2025-02-18:
Hi,

is this Adapter supported or only those listed?
https://www.scantool.net/downloads/updates/obdlink_mxp/

Wondering if this could also work with K-Line on VAG.

Thanks

closed 2025-02-18 by uholeschak

einfacharthur, 2025-02-18:
> If the adapter an correct ELM327 implementation it should work, but not for K-Lin and VAG. https://github.com/uholeschak/ediabaslib/blob/master/docs/AdapterTypes.md

> Recommended ELM327 versions are 1.4b, 1.5 and origin 2.1, which are based on PIC18F25K80 processor (no MCP2515 chip).
ELM 327 v1.5 does not exist if i read correctly on this Wikipedia Article?
https://en.wikipedia.org/wiki/ELM327

I ordered this one and will try to flash it: https://de.aliexpress.com/item/1005005954955254.html

My OBDLink MX+ supports this ELM327 Version Features:

[image]

uholeschak, 2025-02-18:
This check means nothing, they simply report some features. If they are correctly implemented the tool is unable to detect.
I expect this is no PIC processor but ARM.