uholeschak/ediabaslib #204: increasing baudrate on an FTDI cable

whatotter, 2025-05-13:
D-can supports 500kbps (iirc?) which is pretty quick, although this library only uses 9600 baud (using an FTDI cable), which is slower than k-line

https://github.com/uholeschak/ediabaslib/blob/25ebb91a2e7a94fa266ace04745c6dae4063de5b/EdiabasLib/EdiabasLib/EdFtdiInterface.cs#L163

is there any sort of way to increase the baudrate, other than changing the code itself?  
am i stupid and missed over something?

uholeschak, 2025-05-13:
This class is not used any more and could be removed. It's existing only for historical reasons.

closed 2025-05-13 by uholeschak

whatotter, 2025-05-13:
thank you for your extremely quick response, very much appreciated :)

i see what you mean, i guess `EdInterfaceObd.cs` is what i was looking for

https://github.com/uholeschak/ediabaslib/blob/3a8b5fd5e9b985de53fb5d7fb3d3f4cb62938a38/EdiabasLib/EdiabasLib/EdInterfaceObd.cs#L799

i will test faster baudrates just for fun 🙃