uholeschak/ediabaslib #205: K-line mode/KWP1281 protocol

mariustoporascu, 2025-05-20:
I finally got the preprogrammed adapter and did some tests on my Seat Ibiza 6l 2008(1,4l R3 EDC)
EEPROM INFO(got from other software):
bosch partnumber: 0281012708
software id: 045906019BQ SG7627
engine type: BNV (80 bhp/195 Nm)
ecu type: EDC15P++
checksum type: VAG EDC15P V4.1 2002+ ChecksumOK
The reason i'm here is that the reading of engine tables (mot1281) is quite slow.
After configuration is done, when i go to main activity and i connect, the refreshing of the live data is also slow.
With VCDS this is not the case.
So i'm thinking that it might have something to do with timeout defaults for kwp1281, since i've noticed that there are some vars for this in the files.
Could you tell me from your testing experience if this is related to the custom firmware capabilities at the moment, or the car ecu is just lazy if multiple tables are set for live data or tested for availability during initialization(configuration process).
I have tested on a kwp 2000 car(GOLF 5 PLUS) aswell and the readings were fast. That's why i'm thinking that it might have something to do with the kwp1281 reading/testing process.
If you can point me in the right direction i could try to test/improve the kwp1281 defaults.
Best regards

uholeschak, 2025-05-20:
KWP1281 ist a very slow protocol, so this is expected. When changing timeouts it will probably not work any more with other ECUs,

closed 2025-05-20 by uholeschak

mariustoporascu, 2025-05-20:
I understad it's a slow protocol, but for example in vcds you can load 3 measuring groups and it can read the data at a decent speed.
What's more disturbing is that scanning tables for ecu takes a very long time(about 15 minutes for engine module). That's why I came here to ask if the timeout for checking if a specific module has a particular measuring block is within the adapter itself or within the mobile app. For example the seat ibiza has only 16 measuring blocks discovered(which i don't mind) but the wait for 15 minutes it's a pain. :)

uholeschak, 2025-05-20:
You should check the trace files for further investigation. I would expect that there are multiple disconnects or there are simply communication problems and it ends in reconnects.

mariustoporascu, 2025-05-21:
The connections are fine. Anyway. Wanted to ask how to get the licence for bmw vehicle? It was advertised as included in pre programmed device :D

uholeschak, 2025-05-21:
At the moment there is no license required.

uholeschak, 2025-05-22:
In the trace I could only see:
Resolving VAG part number: '045906019BQ', HW part number: '------------', Address: 01
Basically I need your generated configuration and the complete trace of the ECU detection process in the configuration generator.
The traces must contain all required data to simulate the vehicle.
The ECU type is detected from the part number by an obscure logic. Maybe simply the ECU file is missing in the database and this is a fallback.

uholeschak, 2025-05-22:
I don't know if the last trace corresponds to the correct ECU. The ECU type is unknown.

mariustoporascu, 2025-05-24:
ok i finally got it how the flow for automatic configuration works. it scans all the 255 groups and then it tries to match to a .xml file based on the responses(i guess). but that's kind of inefective if you ask me. I would give to user the option to select the .xml based on the info he knows about his car, and if he doesn't, then perform automatic scan, this could save a lot of time in generating configuration.
Attached the trace file for configuration process, but when i tried to move the configuration folder it got lost somehow.
As a note i have to use OldVagMode = On in order to be able to scan the groups, but other functions like Login,Adaptation,etc works only if OldVagMode = Off. 
Another bug/deffective thing i found is in Adaptation, where if i try to store Group 0, i am getting failure, even if i login to the module(12233 login code).  As a matter of fact i could only read group 2 in adaptation(1, 3, 4, are available in VDCS). Perhaps it has something to do with the login token(i am just assuming, i might be wrong).

[ifh.zip](https://github.com/user-attachments/files/20421729/ifh.zip)

mariustoporascu, 2025-05-24:
I created a python script which outputs the mwdata json attribute in order to bypass the scanning process:
import xml.etree.ElementTree as ET
import base64

def process_xml_file(file_path):
    try:
        # Parse the XML file
        tree = ET.parse(file_path)
        root = tree.getroot()
        
        results = []
        
        # Iterate through all Messwert elements
        for messwert in root.findall('Messwert'):
            # Extract required fields
            ag = messwert.find('AG').text if messwert.find('AG') is not None else ''
            af = messwert.find('AF').text if messwert.find('AF') is not None else ''
            
            # Determine dataType based on SWF
            swf = messwert.find('SWF').text if messwert.find('SWF') is not None else ''
            if swf == 'R':
                data_type = 'real'
            elif swf == 'T':
                data_type = 'string'
            else:
                data_type = 'integer'
            
            # Process SWE for dataInfo
            swe = messwert.find('SWE').text if messwert.find('SWE') is not None else ''
            data_info = base64.b64encode(swe.encode('utf-8')).decode('utf-8') if swe else ''
            
            # Add to results collection
            results.append(f"{ag},{af},{data_type},{data_info}")
        
        # Join all results with semicolon
        return ";".join(results)
    
    except Exception as e:
        print(f"Error processing XML file: {e}")
        return ""

# Example usage
if __name__ == "__main__":
    xml_file_path = "ecu.xml"  # Replace with your XML file path
    output = process_xml_file(xml_file_path)
    print(output)
this way i can get the tables in a second

mariustoporascu, 2025-05-24:
I noticed that OldVagMode is using VAS-PC data, that's not what i meant. Disabling it will stop modules which have kwp1281 from working(when i scan my seat with oldvagmode disabled, no tables got detected). As for the discussion being useless, i just wanted to provide some feedback on my experience with the product(adapter and apps). Maybe if you want, i can provide a scan trace on my seat with OldVagMode disabled if needed for debugging purposes.

uholeschak, 2025-05-26:
Thank I will check it when I have some time ...

mariustoporascu, 2025-05-29:
i managed to get tables for my ecu, the reason was that english vag files misses 045-906-019-bnm.ldat
it is pressent in de files
.

uholeschak, 2025-05-29:
You mean that this is only working with german setting?
I think at the moment the fallback is always english. Maybe I could try all other langues if no en file is present ...

mariustoporascu, 2025-05-29:
Yes, that's correct. I did some debugging to understand how the detection for ecu works. And i stunbled on a match 045-906-019-bnm.ldat. when i hit search in the top level vag dir, it found it in de folder but not in en. I went to the car to test, and everything worked. Except 1 thing. I cannot reset my adaptations, but that's another story. And adaptation for ch 3 doesn't read either(i am trying to reset my maf airflow setting). But i will try to improve the ldat file and see if it will work, i noticed that there are some channels described there(a002,1,info,a002,2,nextinfo, etc)

mariustoporascu, 2025-05-30:
New info. 
I managed to read and test the ch 3 adaptation using edibastool function for the ecu.
For read i used:
Anpassung Lesen with command:
3;WertEinmalLesen
For test i used:
Anpassung Testen with command:
3;32768;WertEinmalLesen
When i try to store the value with:
Anpassung Speichern
3;32768;WertEinmalLesen
I get a NACK or Block bas1281.prg missing.

mariustoporascu, 2025-05-30:
This is what i found exploring traces when running the cmds using Adaptation menu for ecu, and i just used them manually in ediabas tool menu

uholeschak, 2025-05-31:
Have you uploaded also a trace from the adaption process? At least I did not see any data.

mariustoporascu, 2025-05-31:
Uploaded now. Attaching here the config for de language too.
[image]

mariustoporascu, 2025-05-31:
[image]

uholeschak, 2025-06-01:
What I have changed now:

- If the ldat file is not found with the current langue, it will try all other langues
- At the Adaption and Login page: Automatic reconnect after communication failure. Basically KWP1281 communication is structurally unstable so it could disconnect at any time,. Simply try again and press actions as fast as possible!
- The SGBD not found problem is not reproducible. I expect there is a problem with your device. Login is basically working.

uholeschak, 2025-06-01:
Google internal tests require at least one day. So simply wait ...

mariustoporascu, 2025-06-03:
Offtopic. Are you kind to provide a way for direct contact? I got a programmer pickit and i did not know how to stop providing power from pickit side. I burned the obd adapter, the pickit and my pc motherboard. Wanted to ask if by any chance i will be able to salvage any of the pickit or motherboard by taking them to repair shop. Just asking from your experience and knowledge. Thx

uholeschak, 2025-06-03:
I don't know what you problem is, power is disabled by default:
[image]
Also if enabled this will definitely not destroy your mainboard.
There must be another general problem like mass loops ...

mariustoporascu, 2025-06-04:
So i should connect only the nr 16 pin from the obd side? Because i connected the pins 4&5 aswell.

mariustoporascu, 2025-06-05:
Oky will try that when i get my hands on new hardware. Atm i had to replace motherboard in pc :)) l. So to all out there, proceed with caution if u don't want to spend more money that what u expect.
Quick question. Is the custom elm327 firmware able to run ms can(pins 3 and 11)? That is the only reason i wanted to build an adapter myself. Cuz i have the adapter from u for other cars. Just wanted 1 for ford/volvo etc, because ur firmwares are much faster that originals. :d

uholeschak, 2025-06-05:
I don't really understand what you mean. CAN ist on pins 6 and 11, what is supported.
The J1850 bus on pins 2 and 10 is not supported.

mariustoporascu, 2025-06-05:
Ford/volvo and some other brands have 2 can speeds. High speed (pin 6 and 14) and medium speed(pin 3 and 11). Was planing to use your elm327 compatible fimware Because it's more stable And to build a custom adapter with micro switch on-on to comute between high and low speed.

uholeschak, 2025-06-05:
Ok, then this is no standard ELM hardware. If course you could modify software and hardware if you want o to do so ...

uholeschak, 2025-06-13:
Great, sounds good!